EverythingSmartHome/firmware #1: could you kindly confirm pinout for I2C

zala143, 2023-12-17:
Could you kindly confirm the pinout for I2C connection for the EST-PoE-32.  I would also be awesome if you could create some hat variations for the EST-PoE-32 - one that adds a STEMMA QT connection for east hookup to I2C devices from Adafruit etc - thanks

EverythingSmartHome, 2023-12-18:
Hello, you can use any GPIO pins (that aren't input only) for i2c so long as you define them in your code correctly.

Hope that helps

zala143, 2023-12-18:
ok after a bit of testing, I was able to get i2c working using the following - for some reason kept getting an error but forum post somewhere suggested going a clean build and that worked... all good.. thanks - love this little board... I hope you're planning a bunch of hat options...
i2c:
  - id: bus_a
    sda: GPIO13
    scl: GPIO16
    scan: true

EverythingSmartHome, 2023-12-18:
Glad to hear you got it working and thanks for the feedback on the board. We are indeed planning some expansion HATs

closed 2023-12-18 by zala143