EverythingSmartHome/firmware #2: are there any 3D printable cases for the EST-PoE-32

zala143, 2023-12-18:
hopefully logging this type of question here is the correct action, please let me know if not..

are there any 3d case options i can print for the EST-PoE-32 - I see the original artwork has changed on the website - with pictures of the device in a case now gone... happy to buy files if you're selling those too, regards

EverythingSmartHome, 2023-12-18:
The case was never an image on the shop and they haven't changed since launch (unless I'm misremembering but pretty sure they weren't), you are maybe thinking of an image I posted on Twitter, YouTube or Discord?

Anyways that's not important, yes the case is going to be made available, there are a couple of things that need adjusting and then I'll list it on printables

zala143, 2023-12-18:
great - yes maybe it was twitter now you say that.. https://twitter.com/EverySmartHome/status/1726605185018200377/photo/1

I think seeing the case, reading about the board etc was where my brain cogs started to turn - I've been wanting to  standardize all my esp32 hacks into a few board options, the concept you're dreaming up here with hats etc is so good... I would love it if there was a generic one we could use to create our own PCB designs too..

closed 2023-12-18 by zala143

EverythingSmartHome, 2023-12-18:
I really appreciate that and yes this is definitely a board we would love to create expansion boards for!

EverythingSmartHome, 2023-12-20:
Case now uploaded here: https://www.printables.com/model/687849-est-poe-32-esp32-poe-devkit-case-enclosure

Would like to hear the feedback!

zala143, 2023-12-21:
thats awesome Lewis - I will be unwrapping a new bambu labs printer from Santa in a couple of days - your case will be my first print.. I'll leave my flagging Enda 3 to just benchy's here on out :) happy Xmas, 2023 has been a great year for you!!!

zala143, 2023-12-25:
[image]
first print on Bambu P1S :)

EverythingSmartHome, 2023-12-26:
@zala143 Awesome, how did it turn out!? Enjoy the printer!