EverythingSmartHome/firmware #3: Error during Compiling

lightsabata, 2024-01-03:
Hi !
i've just receive a ESP32-C6-DevKit. as i read i understand that i need to compile the firmware manually.
i install esphome on my windows machine download the provided template from the repo, then use the command :
```
esphome run est-esp32-c6-alll-gpios.yaml
```
Unfortunately le compiling seems to have a problem and failed the compiling.
here's the log:

```
INFO ESPHome 2023.12.5
INFO Reading configuration est-esp32-c6-alll-gpios.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO8 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting .esphome\build/est-esp32-c6\.pioenvs
INFO Deleting .esphome\build/est-esp32-c6\.piolibdeps
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.2.2
INFO Compiling app...
Processing est-esp32-c6 (board: esp32-c6-devkitc-1; framework: espidf; platform: platformio/espressif32@6.4.0)
---------------------------------------------------------------
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitc-1.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-C6-DevKitC-1
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50101.230828 (5.1.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Improv @ 1.2.3
Building in release mode
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\api_connection.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\api_frame_helper.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\api_pb2.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\api_pb2_service.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\api_server.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\list_entities.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\proto.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\subscribe_state.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\api\user_services.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32\core.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32\gpio.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32\preferences.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble\ble.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble\ble_advertising.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble\ble_uuid.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_2901.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_2902.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_characteristic.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_descriptor.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_server.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_ble_server\ble_service.o
In file included from src/esphome/components/esp32_ble/ble.cpp:7:
src/esphome/components/esp32_ble/ble.cpp: In member function 'bool esphome::esp32_ble::ESP32BLE::ble_setup_()':
C:/Users/baudoin/.platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]
  263 | }
      | ^
src/esphome/components/esp32_ble/ble.cpp:117:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
  117 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/baudoin/.platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_low_tx_pwr' [-Wmissing-field-initializers]
  263 | }
      | ^
src/esphome/components/esp32_ble/ble.cpp:117:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
  117 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/baudoin/.platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:258:58: error: narrowing conversion of 'efuse_hal_chip_revision()' from 'uint32_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} [-Werror=narrowing]
  258 |     .version_num                = efuse_hal_chip_revision(),                            \
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~^~
src/esphome/components/esp32_ble/ble.cpp:117:40: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
  117 |       esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs\est-esp32-c6\src\esphome\components\esp32_improv\esp32_improv_component.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\gpio\switch\gpio_switch.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\improv_base\improv_base.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\improv_serial\improv_serial_component.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\logger\logger.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\md5\md5.o
Compiling .pioenvs\est-esp32-c6\src\esphome\components\mdns\mdns_component.o
cc1plus.exe: some warnings being treated as errors
*** [.pioenvs\est-esp32-c6\src\esphome\components\esp32_ble\ble.o] Error 1
================= [FAILED] Took 17.93 seconds =================

```
is there something that i missed ?

EverythingSmartHome, 2024-01-03:
Just an FYI, you can actually just run this config in the ESPHome add-on in Home Assistant, no need to install it locally.

Can you try removing the esp32_improv section? That shouldn't have been included in the config, my bad. Make sure to clean the cache after changing

lightsabata, 2024-01-03:
Thanks it work with the esp32_improv section

closed 2024-01-03 by lightsabata

EverythingSmartHome, 2024-01-03:
Perfect thanks, I made the adjustment to the config. I also updated the esp-idf version to 5.1.2, tested compiling on my side and working OK!