EverythingSmartHome/firmware #6: Is ESP32-PoE a supported ethernet device for ESPresence?

zala143, 2024-01-07:
any way to confirm if the EST-PoE-32 - ESP32 PoE Development board is a supported Base Station in ESPresence - if so which ethernet profile to select?

I only have one unit and its currently in use so i cannot test directly - would love if this is a supported ethernet type as I can see this as a perfect board to improve ESPresence reliability. 

Thanks

EverythingSmartHome, 2024-01-07:
I haven't ever tried Ethernet on ESPresence, so took a quick look at the code and saw the network profiles, and saw that the ESP32-PoE config is the same.

Installed ESPresence on the EST-PoE-32, selected that profile and works great!

Good to verify it works on this too! I might try to submit a PR to add our board into the config

EverythingSmartHome, 2024-01-07:
Added a PR [here](https://github.com/ESPresense/ESPresense/pull/1186) for it - hopefully I did it correctly and it can be a part of the drop down menu

zala143, 2024-01-08:
thats awesome.. I will order some more boards now, you made my day!!

zala143, 2024-01-08:
Closed with thanks

closed 2024-01-08 by zala143

zala143, 2024-01-10:
@EverythingSmartHome - I was able to install ESPresense onto the one ESP32-PoE board i have (I've order more).... its so good, threw on a couple of additional I2C sensors for Temp/Humidity & Light - going Ethernet for ESPresense changes the experience for me totally - its so much more responsive, loading config/restarts etc... amazing,!!!  wonder if its worth a video of you showing off your POE board running ESPresence  - I cannot see anyone mentioning you can run via ethernet, i only stumbled upon it..   its makes ESPresnse much more useful! cheers M